Duke-BME-Design/Breath_of_fresh_air #13: Contact HeliCal and order machined springs

epvienneau, 2018-02-04:


epvienneau, 2018-02-08:
Sarah can't find the EMST150 device so we can't measure the spring dimensions so there's no point in blindly ordering springs when there are infinitely many to choose from... Harrison is also out of town/not replying to email. We will order springs once we obtain a reference spring.

epvienneau, 2018-02-08:
Websites suggested by steve earp:

https://www.mddionline.com/how-choose-right-springs-your-medical-device
https://www.thespringstore.com/specialty-springs/springs-for-medical-devices.html#
http://www.generalwirespring.com/medical-device-springs.html
http://www.drtempleman.com/springs-by-industry/medical

sphaerobolus, 2018-02-08:
I'm pretty sure I can find an EMST150 in my stash in Chesterfield. Today's a busy day for me, but tomorrow looks good.

epvienneau, 2018-02-08:
Thanks for the devices Matt!

epvienneau, 2018-02-13:
After discussion with Matt brown, I think it is best we stick with wire springs. 
1) machined springs are too expensive
2) I think we can achieve what we want with longer, concentric, wire springs

closed 2018-02-13 by epvienneau