Duke-BME-Design/Breath_of_fresh_air #93: Design and make PCB for flow rate sensing

epvienneau, 2018-03-26:


epvienneau, 2018-04-07:
Chris has not replied, should I ping him again for help making the PCB or should I just do photo exposure? Alternatively I could contract it out but would prefer to not wait a week or so for it.

mlp6, 2018-04-09:
Matt has exclusively been managing this, so I would defer to @sphaerobolus on the best way to proceed (since he also manages the photo-exposure approach!).

epvienneau, 2018-04-09:
Ok, thanks!

sphaerobolus, 2018-04-09:
I don't see your artwork. Have I overlooked it? Could you send it or point me to it? If you haven't already done so, make sure your traces are 20 mil wide or wider, and that no gaps are less than 10 mil. 

Another group is also looking to fabricate a PCB; I'm currently negotiating a time with them, suggesting tomorrow at 11:00, meeting in the Teer basement. Would be convenient for me if we could knock out both boards in one session. 

Depending on the complexity of your board, we'll choose the best path forward--photo-etch or ask to mill.

epvienneau, 2018-04-09:
Looks like neither of us are available for long enough tomorrow. How about we mill. Diana can try going through Chris because she is an ECE student too and has higher priority than I do. Thanks Matt!

closed 2018-04-17 by epvienneau