Duke-BME-Design/Breath_of_fresh_air #102: Divide 24V into 12V

epvienneau, 2018-04-02:


epvienneau, 2018-04-02:
Might need to match impedances...

epvienneau, 2018-04-02:
managed to roughly match impedance of circuit with 560 ohm resistors, but they got very hot. Not sure if this is ok.

epvienneau, 2018-04-02:
is this the best way to divide the voltage @mlp6?

mlp6, 2018-04-02:
What schematic am I looking for?

epvienneau, 2018-04-02:
Sorry I wasn't more specific- I'm testing what it's going to be like when we have 0 to 24V from the wall instead of +12 V and -12V. I tried using two resistors of the same value to split the voltage in half and create -12V and +12V... and I did succeed but the resistors got super hot

mlp6, 2018-04-02:
You are pulling decent current through those resistors with that config.  Might be flirting with their power rating since they aren't designed for that.  Better would be to use a virtual ground voltage regulator.

epvienneau, 2018-04-02:
ah makes sense, didn't know that existed!

sphaerobolus, 2018-04-02:
I have virtual grounds in Teer.

closed 2018-04-04 by epvienneau